TcMenu/IoAbstraction #210: Move mock implementations from src into a test folder

vzahradnik, 2024-09-07:
**Is your feature request related to a problem? Please describe.**
Main library source contains mock implementations of the library. We should move them into test repository or elsewhere, to clearly separate them from the remaining code.

davetcc, 2024-09-15:
We can't really do that because they are used externally to the project (IE in TcMenu and other projects) so they need to be on the path.

vzahradnik, 2024-09-15:
OK, can we keep this issue open and focus on improving the other projects to not use these resources? If you think the code has to stay there and there is no way how to refactor, you can close this issue.

davetcc, 2025-11-02:
Unfortunately, we just can't do this, other libraries rely on this code and creating another library is too much for a couple of files.  There's just too many libraries already to manage.

closed 2025-11-02 by davetcc